tttapa/Control-Surface-Motor-Fader #3: Problems using 8 motorized faders and Midi Interface

Vahen1981, 2021-08-09:
Hi Pieter! I hope you are fine...
First of all, I should say that I'm pretty new in this things, and since many days I'm trying to understand your code for motorized faders. At this point I think I learnt a lot, but there is still a huge part of your code that is too much complex for my understanding. So, please forgive me if my questions are too basic.

I was trying to run 8 motorized faders using two arduinos pro mini for the fader controller, and a mega 2560 pro (the small version of arduino mega) for the midi controller.

I took your sketch "MIDI-Controller.ino" and I modified a bit to run this 8 faders through this two slaves. I choosed the HardwareSerialMIDI_Interface for this project.
Everything was fine, the eight faders were running very good, untill I disconnected the usb from the mega... I don't know why, but when I did this appears that the master doesn`t send anything to the slaves... I am using the rx0 and tx0 pin for the midi circuit, and an external power supply.

Another sintom is that if I disconnect the second slave, I can disconnect the usb from mega without any problem, but obviously just with 4 faders.

Any suggestion about where is the problem?

tttapa, 2021-08-09:
I don't immediately see any problems with what you've described, that should work fine in theory. Could this be a power issue?

You can start debugging by using a simple program to the Mega, e.g. one that just sends a setpoint change over I2C every second or so, this way you can eliminate any MIDI-related issues. 

If you post your code I'll have a quick look at it, maybe I can spot something obvious.

Vahen1981, 2021-08-10:
I think is not a code problem... In fact... this morning I was trying many things and I just upload your code to the mega, exactly as you wrote... the only difference was that I changed the midi interface for a Hardware serial midi...
The behavior was the same... if I use the usb of the mega connected works perfect (obviously with just the first slave), but if I unconnect the usb  the faders doesn't respond and doesn't send any midi message... to weird... I think is a power issue or some connection... as you say... can I share my schematics with you?

Another thing that I realized was that when I disconnect just the SDA line to the second slave, the first slave start to work fine, but obviously not the second.

Vahen1981, 2021-08-10:
Here is my code anyway, thank you pieter!

```#include <Control_Surface.h>
#include <Wire.h>

uint8_t mcu_idx = 0;    // MCU track number
uint8_t fader_idx = 0;  // Index of the fader connected to the slave
const uint8_t slave_addr = 11; // I²C address of slave fader controller
const uint8_t slave_addr2 = 8; 
//USBDebugMIDI_Interface midi;  // MIDI interface to DAW
//HairlessMIDI_Interface hairless;
HardwareSerialMIDI_Interface midiser = {Serial, 115200};
PBValue pb[8] = {{MCU::VOLUME_1},{MCU::VOLUME_2},{MCU::VOLUME_3},{MCU::VOLUME_4},{MCU::VOLUME_5},{MCU::VOLUME_6},{MCU::VOLUME_7},{MCU::VOLUME_8}};   // Receive pitch bend with setpoint
PitchBendSender<10> sender;             // Send pitch bend with actual position
Hysteresis<4, uint16_t, uint16_t> hyst; // Filter the actual position
constexpr bool debug_print = false;     // Print the data from the slave
bool prevTouchedA = false;
bool prevTouchedB = false;
bool prevTouchedC = false;
bool prevTouchedD = false;
bool prevTouchedE = false;
bool prevTouchedF = false;
bool prevTouchedG = false;
bool prevTouchedH = false;
uint16_t faderposCS[8];
uint16_t lastSetpoint1;
uint16_t prevSetpointDAW1 = 0;
uint16_t prevSetpointCS1 = 0;
uint16_t prevLastSetpoint1 = 0;
uint16_t lastSetpoint2;
uint16_t prevSetpointDAW2 = 0;
uint16_t prevSetpointCS2 = 0;
uint16_t prevLastSetpoint2 = 0;
uint16_t lastSetpoint3;
uint16_t prevSetpointDAW3 = 0;
uint16_t prevSetpointCS3 = 0;
uint16_t prevLastSetpoint3 = 0;
uint16_t lastSetpoint4;
uint16_t prevSetpointDAW4 = 0;
uint16_t prevSetpointCS4 = 0;
uint16_t prevLastSetpoint4 = 0;

uint16_t lastSetpoint5;
uint16_t prevSetpointDAW5 = 0;
uint16_t prevSetpointCS5 = 0;
uint16_t prevLastSetpoint5 = 0;
uint16_t lastSetpoint6;
uint16_t prevSetpointDAW6 = 0;
uint16_t prevSetpointCS6 = 0;
uint16_t prevLastSetpoint6 = 0;
uint16_t lastSetpoint7;
uint16_t prevSetpointDAW7 = 0;
uint16_t prevSetpointCS7 = 0;
uint16_t prevLastSetpoint7 = 0;
uint16_t lastSetpoint8;
uint16_t prevSetpointDAW8 = 0;
uint16_t prevSetpointCS8 = 0;
uint16_t prevLastSetpoint8 = 0;

void setup() {
    Wire.begin();                          // Join the I²C bus
    Wire.setClock(400000);                 // Set higher I²C speed
    Control_Surface.begin();               // Initialize everything else
}

void readFromSlave() {
    // Request the position from the slave over I²C
    uint8_t maxlen = 1 + (1 + fader_idx) * 2;
    Wire.requestFrom(slave_addr, maxlen);
    uint8_t buf[maxlen];
[... 350 lines cut ...]

 void sumar(){
    fader_idx++;
    mcu_idx++;
    if(fader_idx == 4){fader_idx = 0; mcu_idx = 0;}
 }

void loop() {
    Control_Surface.loop();
    static Timer<millis> timer = 5;
    // Don't update at full speed, just every couple of milliseconds
    if (timer) {
        readFromSlave();
        sendF1ToSlave();
        sendF2ToSlave();
        sendF3ToSlave();
        sendF4ToSlave();
        readFromSlave2();
        sendF5ToSlave();
        sendF6ToSlave();
        sendF7ToSlave();
        sendF8ToSlave();
        sumar();
    }
}```

tttapa, 2021-08-10:
> `HardwareSerialMIDI_Interface midiser = {Serial, 115200};`

Are you sure this is the correct baud rate? Hardware MIDI with 5-pin DIN connectors use a baud rate of 31250 symbols per second, not 115200.

31250 is the default, so you can just use:
```cpp
HardwareSerialMIDI_Interface midiser {Serial};
```
I didn't look into the rest of the code in detail, but it's in dire need of some arrays, there's too much code duplication. Also, you need one Hysteresis object for each fader.

Vahen1981, 2021-08-10:
Actually the first baud rate that I try was 31250, and the behavior was the same... I changed it to 115200 just trying to fix it... 
I'm aware about the code duplication sorry :( At the begining I was very confused trying to adapt the code to my hadware, but now I will try to optimize and use some arrays to fix it...
I didn't know about hysteresis, I will try that.
Thanks!

tttapa, 2021-08-10:
> I didn't know about hysteresis, I will try that.

It won't solve the problem you're facing right now, it's just a remark.

I'd try changing the setpoint without MIDI as I mentioned in my first reply.

Vahen1981, 2021-08-11:
Thank you! I will try with a simple program sending setpoints and I will let you know!

Vahen1981, 2021-08-11:
Hi Pieter, is definetely not a midi problem, I did a simple program as you suggest and happens the same... when it is connected to the usb works perfect, but if not nothing happens. this is the program that I tried:

```
#include <Wire.h>
#define slave_addr    11
#define slave_addr2   8

uint16_t prevSetpoint[8] = {0,0,0,0,0,0,0,0};

void setup() {
   Wire.begin();                     
   Wire.setClock(400000);               
}

void sendToSlave(uint8_t slaveIdx, uint16_t setpoint){
  for(byte idx = 0 ; idx < 4 ; idx++){
    uint16_t data = setpoint;
    data |= idx << 12;
    Wire.beginTransmission(slaveIdx);
    Wire.write(reinterpret_cast<const uint8_t *>(&data), 2);
    Wire.endTransmission();
    prevSetpoint[idx] = setpoint;
    delay(250);
  }
}

void loop() {
  sendToSlave(slave_addr, 1023);
  sendToSlave(slave_addr2, 1023);
  sendToSlave(slave_addr, 0);
  sendToSlave(slave_addr2, 0);
}
```

So I assume, must be a power issue... or maybe a proble with the mega... I will keep researching...

Vahen1981, 2021-08-12:
I finally solve it!! I'm so happy haha
Apparentely the midi controller should begin after the fader controller... I just put a delay(1000) in the setup of the midi controller, before the Wire.begin() and now is working perfect without the usb connection!

tttapa, 2021-08-12:
Glad to hear! 

There seems to be a timeout option in the Wire library: https://github.com/arduino/Arduino/issues/10803  
I don't have much time right now, but it might be interesting to look into to prevent issues like this one.

Vahen1981, 2021-08-12:
I'll check that, thank you for your support!

moimeme81, 2024-01-31:
@Vahen1981 since you got it to work, it would be interesting if you could create a fork with your code, wiring and all. At the very least it would be easier to follow that a code in a comment of an issue...