tttapa/Control-Surface-Motor-Fader #7: Sometimes a fader goes crazy, after hours working

Vahen1981, 2021-11-23:
Hi Pieter,
After many months working I finally can say my motorized faders are working ok, almost perfect... but I detected this issue when I was working in a musical production, mixing with cubase after 3 o 4 hours using my control surface with motorized faders... suddenly one of the faders goes crazy, it goes to the lower position and no matter what I do, the fader stay there. I move the virtual fader in the DAW and nothing happens, I move the real fader and the motor stops when is touched, so I can move it, but after release it, the fader goes again to the zero position. And the most worrying thing, is that the motor fader doesn´t stop when it is in the lower position, then it starts to heat up... very hot... So I am afraid that one day I will forget to turn it off, and this problem will occur with the consequence that the case may burn or a worse accident...

Time before I had similar problems whith all the faders, and with just after 20 minutes working... and I fix it making a pcb and connecting all the controllers (arduino pro mini) near between them... I think it was a I2C problem, because before the cables were too long... So I assume I fix almost everything about the I2C communication problem, but anyway in some moments the problem persist...

Do you have some suggestion?

The Control surface has 9 faders (8 channels and master), and 4 arduino pro-mini (3 for motor control, and master for midi interface). I use it 4.7k pull-up resistors for SDA and SCL lines. Do you think that value is ok?

Thanks!

tttapa, 2021-11-23:
If the I2C fails and returns 0xFF for every byte you read, that would overflow [here](https://github.com/tttapa/Control-Surface-Motor-Fader/blob/9b20111e7eb25f09a17da3a1a4eb6438c0d54247/Motor-Controller/Controller.hpp#L68)
essentially resulting in a negative setpoint.

Stronger pull-ups (you could probably go as low as 470Ω) and short wires might help, as well as reducing the I2C clock frequency.  
However, you might want to add some kind of sanity check in the code before you change the setpoint. You know the range of the setpoint values, so if you get 0xFFFF, you know something's up. I think newer versions of the Wire library have some error handling features that might be worth looking into. The problem is that once the TwoWire peripheral encounters an error (e.g. if it thinks another master has claimed the bus), it essentially locks up until you explicitly reset it.

Vahen1981, 2021-11-24:
Thank you Pieter, I will try with other values for the resistor and also reducing the clock frequency, at the moment I was using 400000...
I will test that, and also find more information about the wire library. Thanks!

huykaamaa, 2023-08-30:
I have same problem, but fader suddenly go to max position. I try remove pro micro, only use hairless midi on nano but it still error.
I check serial monior of nano it show 800(depending on daw)        1023      1022
And motor get hot, when i press reset on nano everything ok
I think problem is I set Ki too high (>100 problem occurs more often

commit 7fda1169 (2023-09-03): Fix #7
closed 2023-09-03 by tttapa