tttapa/Control-Surface-Motor-Fader #9: Use on Arduino Uno

mediaMicha, 2022-02-06:
Hello,

I'm not able to compile the .cpp at Arduino development. Can you please make it easy for me, and upload the src directly for Arduino?

Thank you. 

Michael

tttapa, 2022-02-06:
1. Download the .ZIP file using the green <kbd>Code</kbd>  button on the home page of this repository
2. Extract it to a convenient location on your computer
3. Open the Arduino IDE
4. Use <kbd>Ctrl+O</kbd> or use the `File > Open ...` menu
5. Browse to the `Control-Surface-Motor-Fader-master/Motor-Controller` folder you just extracted and open the `Motor-Controller.ino` file
6. Select the Arduino UNO in the `Tools > Board` menu and select the correct port
7. Use <kbd>Ctrl+U</kbd> to compile and upload the code to the Arduino

mediaMicha, 2022-02-07:
Thank you, for this very fast answer. But it's missed some files:

In file included from sketch/main.cpp:2:0:
ADC.hpp:6:10: fatal error: Arduino_Helpers.h: No such file or directory
 #include <Arduino_Helpers.h> // EMA.hpp
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Arduino_Helpers.h: No such file or directory

And:

In file included from sketch/main.cpp:2:0:
ADC.hpp:8:10: fatal error: AH/Filters/EMA.hpp: No such file or directory
 #include <AH/Filters/EMA.hpp> // EMA filter
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
AH/Filters/EMA.hpp: No such file or directory

I download this files/folder from your GitHub, I'm not sure, in with folder it to copy. I thought to folder: libraries, but, it don't find the files.

mediaMicha, 2022-02-07:
Thank you, this was the goal. Now I have it on the board. Now I must look, how it's work.

closed 2022-02-09 by mediaMicha