tttapa/Control-Surface-Motor-Fader #21: Using a TB6612FNG

Ilosariph, 2024-01-16:
Hello Pieter,
I have tried implementing very basic controls for a motorized fader myself. I can't get it to stop on point like your video shows. I then saw your website, and wanted to try your library, as it looks really thought out. I already have a Sparkfun TB6612FNG (https://www.sparkfun.com/products/14451) and you mentioned the TB6612FNG (I know you meant the chip, and I can't read datasheets, but I guess it would still have more connections than a L293D) in your hardware documentation. I have one small problem though:
According to the schematic PJordan2820 made in https://github.com/tttapa/Control-Surface-Motor-Fader/issues/6, the L293D only has two serial connections to the Arduino. The TB6612FNG I am working with has 4 (AIN1, AIN2, PWMA and Standby). How would I connect that to the Arduino to make it work with your library?
Thanks for taking the time to answer (or at least reading the question)

tttapa, 2024-01-16:
Because of the asymmetry of this driver (see the H-SW Control Function table on page 4 of the [datasheet](https://www.sparkfun.com/datasheets/Robotics/TB6612FNG.pdf), HHHH→LL and LLHH→Off), you should use the modifications explained in https://github.com/tttapa/Control-Surface-Motor-Fader/issues/18#issuecomment-1670147446

Using those changes, then connect the driver as follows.

| Arduino pin | Driver pin |
|:---|:---:|
| 2 (PD2) | AIN1 |
| 3 (PD3/OC2B) | PWMA |
| 4 (PD4) | AIN2 |
| 5V (Vcc) | STBY |

To wire the motor and the power supply, see the Typical Application Diagram on page 7 of the datasheet.

If you want to use more than one fader, you'll need to change the code for the other motors as well.

Ilosariph, 2024-01-16:
Thanks for that quick answer. The code change seems simple enough and makes sense (though I don't understand most of the code, I understand what you are changing). I'll try it tomorrow and let you know the results of my attempt.

Ilosariph, 2024-01-17:
So I've tried it (with the test reference), but it's not working. I have a theory:
I am using an Arduino Mega rev3. The [pinout](https://content.arduino.cc/assets/Pinout-Mega2560rev3_latest.pdf) of it shows the PD2 as D19, the OC2B (which is the important part about this connection if I understood your comment on the other issue correctly) is D9 and PD4 is the RX LED, which makes it inaccessible for me.
I did try it with AIN1 to D19, PWMA to D9 and AIN2 to D3 (for a lack of better option). I tried using D2, D3 and D4 as well.
Reading the fader position with the `print_controller_signals` flag works as expected. Driving the motor and touch detection work with a simple Arduino script, so the fader or driver aren't broken and the power supply should be sufficient.
Is there an easy way to make this work with the Arduino mega or should I just buy a nano?

Hardware:
The [fader](https://www.reichelt.de/it/de/schiebepotentiometer-mono-5-kohm-linear-rs60n11m9-lin5k-p73883.html?&trstct=pol_3&nbc=1) is the 5K version of the one you used.
The connections on the diagram are made with the letters / numbers from it's datasheet.
[image]

tttapa, 2024-01-18:
It should be possible to make it work on the ATmega2560, but you'll have to change the pins used.

Specifically:
- pick a different port for the touch pins (e.g. PA instead of PB)
- pick different motor AIN1/2 pins
- change the overrun indicator pin from PB5 to PB7

In the code, change the following lines:

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/main.cpp#L146-L148

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/main.cpp#L385

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/main.cpp#L408

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/Touch.hpp#L53-L54

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/Touch.hpp#L78-L82

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/Motor.hpp#L64-L65

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/Motor.hpp#L98

https://github.com/tttapa/Control-Surface-Motor-Fader/blob/fe8a54f912ea0bac2d75647e6f40e23807e00823/Motor-Controller/Motor.hpp#L137

(Or your modified version)