tttapa/Control-Surface-Motor-Fader #26: Using it with ESP32-S2

labodezao, 2025-01-17:
Hello,

I'm not able to compile the .ino Arduino project on esp32-s2. I use the esp32 library and it seems that the library control surface is not seen when I turn the board to esp32S2

Do you have any idea ?

Thank you.

tttapa, 2025-01-18:
The motor controller sketch is hand-optimized for the ATmega328P, so it is not compatible with the ESP32-S2.

The MIDI controller sketch should work on the ESP32-S2, though. Could you provide more details about the steps you've followed and the exact errors you're getting?

labodezao, 2025-01-18:
I need a motor fader interface for esp32 S2, how can I adapt you code to
make the things working ?

---------------------------------------------------------------
http://labodezao.fr/
---------------------------------------------------------------
Ewen d'Aviau
9 rue fernand de Magellan
44600 Saint-Nazaire
06-02-37-53-22

tttapa, 2025-01-18:
Most of the concepts are transferable, but you'd need to completely rewrite all the code related to the hardware timers, ADC and GPIO direct port manipulation. This will require detailed knowledge of the ESP32-S2's timer and ADC peripherals.

- https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/adc_continuous.html
 - https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/touch_pad.html
 - https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/touch_element.html
 - https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/ledc.html.

You'll probably have to use the LED control timers, because the ESP32-S2 lacks the MCPWM peripheral that's included in some other ESP32 variants.

labodezao, 2025-01-19:
Could you help me a little bit if i have troubles on the pathway ?
I am engineer but microcontrollers are a bit far away now !

---------------------------------------------------------------
http://labodezao.fr/
---------------------------------------------------------------
Ewen d'Aviau
9 rue fernand de Magellan
44600 Saint-Nazaire
06-02-37-53-22

labodezao, 2025-01-19:
> Sure, feel free to ask your questions here, but I can't promise much. I
> don't have much free time, unfortunately, and I've never used the ESP32-S2.

Would it be more easy on a standart esp32 board ?
I just need 2 Faders so 2 Adc is sufficent + wifi interface

---------------------------------------------------------------
http://labodezao.fr/
---------------------------------------------------------------
Ewen d'Aviau
9 rue fernand de Magellan
44600 Saint-Nazaire
06-02-37-53-22

labodezao, 2025-01-19:
The first error i get when i try to comile over esp32 arduino library : 

In file included from D:\Downloads\Control-Surface-Motor-Fader-master\Control-Surface-Motor-Fader-master\Motor-Controller\ADC.hpp:2,
                 from D:\Downloads\Control-Surface-Motor-Fader-master\Control-Surface-Motor-Fader-master\Motor-Controller\main.cpp:2:
D:\Downloads\Control-Surface-Motor-Fader-master\Control-Surface-Motor-Fader-master\Motor-Controller\Registers.hpp:3:10: fatal error: avr/io.h: No such file or directory
    3 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: avr/io.h: No such file or directory

labodezao, 2025-01-19:
Do you think it could be another option to code this using micropython ?
I have a lot more of experience on micropython

tttapa, 2025-01-19:
> fatal error: avr/io.h: No such file or directory

AVR is the architecture of the ATmega328P microcontroller that the code was originally designed for. If you want to port it to the ESP32-S2 microcontroller, the hardware peripheral configuration will be completely different, and you won't be able to use the AVR-specific registers (such as the ones defined in `<avr/io.h>`). You'll have to rewrite all the AVR-specific code, there are no shortcuts here.

> That why i don't need like i2c, and most of the features.

I2C etc. is the easy part, as it uses the high-level Arduino Wire API. The actual control loop, ADC, timers etc. are much more low-level and require manual porting.

> Do you think it could be another option to code this using micropython ?

For this kind of low-level code, I see no benefits in using micropython. The most important aspect of the motor controller is to have a hardware timer to trigger the ADC measurements and to have the control loop run at a regular rate. I don't know if micropython even gives you sufficient low-level control to configure a setup like that. The top results on Google (e.g. https://forum.micropython.org/viewtopic.php?t=604) don't seem very promising.

> I just need 2 Faders so 2 Adc is sufficent + wifi interface

One of the ESP32-S2's ADCs is shared with the WiFi module, so you'll have to multiplex the remaining ADC.

labodezao, 2025-01-19:
I find this in the exemples for S2 : 

```

/*
 Repeat timer example

 This example shows how to use hardware timer in ESP32. The timer calls onTimer
 function every second. The timer can be stopped with button attached to PIN 0
 (IO0).

 This example code is in the public domain.
 */

// Stop button is attached to PIN 0 (IO0)
#define BTN_STOP_ALARM 0

hw_timer_t *timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile uint32_t isrCounter = 0;
volatile uint32_t lastIsrAt = 0;

void ARDUINO_ISR_ATTR onTimer() {
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  isrCounter = isrCounter + 1;
  lastIsrAt = millis();
  portEXIT_CRITICAL_ISR(&timerMux);
  // Give a semaphore that we can check in the loop
  xSemaphoreGiveFromISR(timerSemaphore, NULL);
  // It is safe to use digitalRead/Write here if you want to toggle an output
}

void setup() {
  Serial.begin(115200);

  // Set BTN_STOP_ALARM to input mode
  pinMode(BTN_STOP_ALARM, INPUT_PULLUP);

  // Create semaphore to inform us when the timer has fired
  timerSemaphore = xSemaphoreCreateBinary();

  // Set timer frequency to 1Mhz
  timer = timerBegin(1000000);

  // Attach onTimer function to our timer.
  timerAttachInterrupt(timer, &onTimer);

  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter) with unlimited count = 0 (fourth parameter).
  timerAlarm(timer, 1000000, true, 0);
}

void loop() {
  // If Timer has fired
  if (xSemaphoreTake(timerSemaphore, 0) == pdTRUE) {
    uint32_t isrCount = 0, isrTime = 0;
    // Read the interrupt count and time
    portENTER_CRITICAL(&timerMux);
    isrCount = isrCounter;
    isrTime = lastIsrAt;
    portEXIT_CRITICAL(&timerMux);
    // Print it
    Serial.print("onTimer no. ");
    Serial.print(isrCount);
    Serial.print(" at ");
    Serial.print(isrTime);
    Serial.println(" ms");
  }
  // If button is pressed
  if (digitalRead(BTN_STOP_ALARM) == LOW) {
    // If timer is still running
    if (timer) {
      // Stop and free timer
      timerEnd(timer);
      timer = NULL;
    }
  }
}
```
Could that be usefull ?

labodezao, 2025-01-19:
ok, With my friend GPT, i made this code and it's working on the board:
```

#include <Arduino.h>
#include "driver/ledc.h"
#include "driver/adc.h"

// Définir les constantes de présélection de l'ADC et des minuteries
#define ADC_WIDTH ADC_WIDTH_BIT_13  // 13 bits est souvent plus compatible avec la version 3.1.1
#define ADC_CHANNEL ADC1_CHANNEL_0  // Par exemple, utiliser le canal 0
#define TIMER_PRESCALER 1000  // Exemple de valeur pour la fréquence du timer

// Configuration du Timer 0
void setTimer0Prescaler(uint32_t prescaler) {
    ledc_timer_config_t timer_config = {
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .duty_resolution = LEDC_TIMER_8_BIT,  // 8 bits de résolution pour le timer
        .timer_num = LEDC_TIMER_0,
        .freq_hz = prescaler,  // Fréquence du timer
        .clk_cfg = LEDC_AUTO_CLK
    };
    ledc_timer_config(&timer_config);
}

// Configuration de l'ADC
void setADCPrescaler(uint8_t prescaler) {
    // Configuration de l'ADC avec la largeur appropriée (13 bits)
    adc1_config_width(ADC_WIDTH);  // Configure l'ADC avec une largeur de 13 bits (ou 12 si disponible)
    adc1_config_channel_atten(ADC_CHANNEL, ADC_ATTEN_DB_0);  // Configure le canal 0 de l'ADC
}

// Fonction d'initialisation
void setup() {
    Serial.begin(115200);  // Initialisation du port série pour le débogage

    // Initialiser les minuteries
    setTimer0Prescaler(TIMER_PRESCALER);

    // Initialiser l'ADC
    setADCPrescaler(13);  // Exemple de présélection

    // Autres initialisations, si nécessaire
    Serial.println("Configuration terminée");
}

// Boucle principale
void loop() {
    // Exemple de lecture de l'ADC
    int adc_value = adc1_get_raw(ADC_CHANNEL);
    Serial.print("Valeur ADC : ");
    Serial.println(adc_value);

    delay(1000);  // Attente de 1 seconde avant la prochaine lecture
}

```

Am I on the right pathway ?

tttapa, 2025-01-19:
> Am I on the right pathway ?

Hard to say, I'm not familiar enough with the ESP-IDF APIs to proofread your code. That said, experience taught me that ChatGPT is really bad at this kind of embedded, library-specific code, and it will just make up all kinds of functions that don't actually exist. In the code you posted, the function names don't make sense, they don't actually use the prescaler value at all, and they don't actually _do_ anything (the functions might be configuring some of the peripherals, but the code doesn't actually read any ADC values, it doesn't actually output any PWM waveforms, and it doesn't actually handle any timer events).

I'm afraid I won't be able to help with the actual ESP32-S2 code, you may have more luck with that on e.g. the Arduino Forum. If you want to discuss the design of the motor controller code in this repository (which is documented here: https://tttapa.github.io/Pages/Arduino/Control-Theory/Motor-Fader/Architecture.html), you're of course free to ask questions about that in this issue tracker.